leonboe1/GoogleFindMyTools #2: Running on local workstation

MarcoGhise, 2024-12-06:
Hi Leon.
First of all, thank you for sharing this solution, I attempted to run this from my local workstation but I got this error after entering my Google account:

`  File "\GoogleFindMyTools\.venv\Lib\site-packages\aiohttp\helpers.py", line 671, in __exit__
    raise asyncio.TimeoutError from exc_val
TimeoutError`

Is this solution intended to run only in Android environment?
Generally speaking, can you provide a kick-off to configure and run the project?

Thanks in advance for your time

leonboe1, 2024-12-06:
Hi there, 

the project does not need to run in an Android environment. However, I only tested it with macOS so far. Do you use Pycharm? What is the last thing you see (in browser, in terminal) before this error appears? Did you try to start main() again?

I might add detailed instructions later. Right now, the project is still very much work in progress.

Best,
Leon

MarcoGhise, 2024-12-06:
Thanks for your time Leon.

I run the application from Pycharm in Windows OS.

This is my complete Run console

`
C:\Users\admin\AppData\Local\Programs\Python\Python312\python.exe C:\develop\GoogleFindMyTools\main.py 
[UsernameProvider] Username was not setup yet. Enter your Google Username (Email without @gmail.com)') and hit enter: usermine

[AppTokenRetrieval] Asking Server for android_device_manager token.
[TokenCache] Using cached username
Error during gcm checkin post attempt 1 out of 3
Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\client.py", line 701, in _request
    conn = await self._connector.connect(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\connector.py", line 544, in connect
    proto = await self._create_connection(req, traces, timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\connector.py", line 1050, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\connector.py", line 1363, in _create_direct_connection
    transp, proto = await self._wrap_create_connection(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\connector.py", line 1109, in _wrap_create_connection
    sock = await aiohappyeyeballs.start_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohappyeyeballs\impl.py", line 89, in start_connection
    sock, _, _ = await _staggered.staggered_race(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohappyeyeballs\_staggered.py", line 160, in staggered_race
    done = await _wait_one(
           ^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohappyeyeballs\_staggered.py", line 41, in _wait_one
    return await wait_next
           ^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\firebase_messaging\fcmregister.py", line 136, in gcm_check_in
    async with self._session.post(
               ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\client.py", line 1423, in __aenter__
    self._resp: _RetType = await self._coro
                           ^^^^^^^^^^^^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\client.py", line 607, in _request
    with timer:
         ^^^^^
  File "C:\Users\admin\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohttp\helpers.py", line 671, in __exit__
    raise asyncio.TimeoutError from exc_val
TimeoutError
`

Basically, just after asking me "Enter your Google Username " it blocked with last message "TimeoutError".
I checked connection to 'https://android.clients.google.com/checkin' and it seems to work.
Have you any ideas about the issue?
Thanks

leonboe1, 2024-12-07:
Thanks for the report. This seems to be a bug with the Firebase Library. I have also noticed that this library is a little bit buggy. The first registration takes some time for me, but it eventually succeeds. I’ll take a look at this issue sometime.

It might also be a issue with your firewall, maybe it is blocking Python to communicate with Google’s server.

MarcoGhise, 2024-12-09:
Thanks for your suggestions, actually I'm working behind a Proxy and, running this endpoint by Postman setting a proxy, it works!
Basically I need to understand how set proxy for aiohttp module, it seems to be ignored when I put this in environment variable.

Anyway, it's an issue outside your project scope, so, I'd close this issue.
Thank you very much for your support.

closed 2024-12-09 by MarcoGhise