leonboe1/GoogleFindMyTools #4: TypeError: Binary Location Must be a String

P6g9YHK6, 2024-12-29:
I am getting the following error, i am not certain that i am running it correctly tho.
also it seems to me that setuptools is missing from the requirements.

```
⚡XXXX ❯❯ & C:/Users/XXXX/AppData/Local/Programs/Python/Python312/python.exe c:/Users/XXXX/GoogleFindMyTools/main.py
[UsernameProvider] Username was not setup yet. Enter your Google Username (Email without @gmail.com)') and hit enter:XXXX.YYYYY          
[AppTokenRetrieval] Asking Server for android_device_manager token.
[TokenCache] Using cached username
GCM register request attempt 1 out of 2 has failed with Error=PHONE_REGISTRATION_ERROR
[FCMReceiver] Credentials updated.
[FCMReceiver] Retrieved FCM Token successfully: ZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZZ
[FCMReceiver] Listening for notifications.
[AuthFlow] This script will now open Google Chrome on your device.
    Make that you allow Python (or PyCharm) to control Chrome (macOS only).
    Please login with your Google account.
[AuthFlow] Press Enter to continue...
[AuthFlow] Installing ChromeDriver...
Traceback (most recent call last):
  File "c:\Users\XXXX\GoogleFindMyTools\main.py", line 9, in <module>
    list_devices()
  File "c:\Users\XXXX\GoogleFindMyTools\NovaApi\ListDevices\nbe_list_devices.py", line 43, in list_devices
    result_hex = request_device_list()
                 ^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\NovaApi\ListDevices\nbe_list_devices.py", line 19, in request_device_list
    result = nova_request(NOVA_LIST_DEVICS_API_SCOPE, hex_payload)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\NovaApi\nova_request.py", line 18, in nova_request
    android_device_manager_oauth_token = get_adm_token(get_username())
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\Auth\adm_token_retrieval.py", line 19, in get_adm_token
    cached_adm_token = request_token(username, "android_device_manager")
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\Auth\token_retrieval.py", line 16, in request_token
    aas_token = get_aas_token()
                ^^^^^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\Auth\aas_token_retrieval.py", line 27, in get_aas_token
    return get_cached_value_or_set('aas_token', _generate_aas_token)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\Auth\token_cache.py", line 18, in get_cached_value_or_set
    value = generator()
            ^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\Auth\aas_token_retrieval.py", line 17, in _generate_aas_token
    token = request_oauth_account_token_flow()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\Users\XXXX\GoogleFindMyTools\Auth\auth_flow.py", line 26, in request_oauth_account_token_flow
    driver = uc.Chrome(options=chrome_options)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 372, in __init__
    options.binary_location = (
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python312\Lib\site-packages\selenium\webdriver\chromium\options.py", line 54, in binary_location
    raise TypeError(self.BINARY_LOCATION_ERROR)
TypeError: Binary Location Must be a String
```

P6g9YHK6, 2024-12-29:
the issue was that chrome was not installed on my device in the way that the library was expecting.
i installed ungoogled-chromium via chocolatey and declared the path found with
```
py
>>> import shutil
>>> print(shutil.which("chrome"))
C:\ProgramData\chocolatey\bin\chrome.EXE
```
set it in the binary variable added in the PR and voila.

leonboe1, 2025-01-02:
Great that it works for you now. I also tried to debug this issue today. Weirdly, for my system it would *not* work if I set the path. Therefore, I only added a console print if Chrome is not detected on the system. In this case, your solution if offered. That's some python library weirdness, haha.

Regarding your project - Not sure what you are trying to accomplish, but so far, the network quality of Google's Find My seems to be lacking. It often seems to drops sightings from other devices for "privacy" reasons.

P6g9YHK6, 2025-02-16:
i have got some compatible devices in my account now 
i still have got the same error but when trying to get the location.
is there some other places where the executable would need to be specified ?

P6g9YHK6, 2025-02-16:
found it

options.binary_location = "C:\\ProgramData\\chocolatey\\bin\\chrome.exe"
to be added in shared_key_flow.py

leonboe1, 2025-02-17:
Good catch. I've also added more explicit error messages for shared_key_flow.py, if Chrome was not found.

P6g9YHK6, 2025-02-17:
i think on first run i would be wise to have a little prompt if chrome is not detected and if not set a custom var for every part of the software for people not to have to hunt for these

and later have a docker container with chrome/vnc packaged into it ;)

leonboe1, 2025-02-17:
Yes, true. I just tried to implement a better Chrome detection on the branch [better-chrome-detection](https://github.com/leonboe1/GoogleFindMyTools/tree/better-chrome-detection). Does this work for you without modifying the source code?

P6g9YHK6, 2025-02-18:
Works first time. 👍👍👍

P6g9YHK6, 2025-02-18:
you cheater 🤣😘

[image]

closed 2025-02-18 by P6g9YHK6