leonboe1/GoogleFindMyTools #8: Unable to Retrieve Location from Tag

Martin22, 2025-02-01:
Retrieving location from the tag works randomly—some days, it works all day, and then suddenly, I receive the error below. Is this a bug, or does Google block it due to frequent queries?

Even when retrieving the tag's location doesn't work, the phone's location is still returned.

```
The following trackers are available:
1. Galaxy S24+: 6683bb68-0000-25a7-9a0a-883d24ffe01c
2. Chipolo ONE Point: 677db4d7-0000-234b-ab3f-14c14efd1b80
Select the tracker number to query: 2
[LocationRequest] Requesting location data for device with canonic ID: 677db4d7-0000-234b-ab3f-14c14efd1b80
[FCMReceiver] Retrieved FCM Token successfully: xxx
[FCMReceiver] Listening for notifications.
[TokenCache] Using cached username
[AdmTokenRetrieval] Using Cached App Token.
[NovaRequest] Status Code: 200
[NovaRequest] Response Hex:
[NovaRequest] Request performed successfully.
[FCMReceiver] Notification received!
[FCMReceiver] Object: {'data': {'com.google.android.apps.adm.FCM_PAYLOAD': 'xxx'}
[FCMReceiver] Notification: 0:1738347649627537%7031b2e6f9fd7ecd
[FCMReceiver] Data Message: None
[FCMReceiver] Decoded FMDN Message: xxx
[<function get_location_data_for_device.<locals>.handle_location_response at 0x7e3665c14c20>]
[FCMReceiver] Calling location update callback.
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[FCMReceiver] Object: {'data': {'com.google.android.apps.adm.FCM_PAYLOAD': xxx'}, 'from': '289722593072', 'priority': 'normal', 'fcmMessageId': '09a3dbaa-c926-479f-945d-bc11992522d4'}
[FCMReceiver] Notification: 0:1738388931066304%7031b2e6f9fd7ecd
[FCMReceiver] Data Message: None
[FCMReceiver] Decoded FMDN Message: xxx
[<function get_location_data_for_device.<locals>.handle_location_response at 0x7e3665c14c20>]
[FCMReceiver] Calling location update callback.
[LocationRequest] Location request successful. Response:
fcmMetadata {
  requestUuid: 6f68b235-0a76-452a-aa9c-475b7060e1cf
}
deviceMetadata {
  identifierInformation {
    type: 2
    canonicIds {
      canonicId {
        id: 677db4d7-0000-234b-ab3f-14c14efd1b80
      }
    }
  }
  information {
    deviceRegistration {
      deviceTypeInformation {
        deviceType: 1
      }
      encryptedKeys {
        encryptedIdentityKey: "xxx"
        keyVersion: 1
        encryptedAccountKey: "xxx"
        keyTimestamp {
          2025-01-10 10:36:12+01:00
        }
        encryptedSha256AccountKeyPublicAddress: "xxx"
      }
      manufacturer: Chipolo
      model: Chipolo ONE Point
    }
    locationInformation {
      reports {
        recentLocationAndNetworkLocations {
          recentLocation {

          }
          recentLocationTimestamp {
            2025-02-01 06:48:51+01:00
          }
        }
      }
    }
    accessInformation {
      email: kukal.m@gmail.com
      hasAccess: True
      isOwner: True
      thisAccount: True
    }
  }
  userDefinedDeviceName: Chipolo ONE Point
  imageInformation {
    imageUrl: https://lh3.googleusercontent.com/tH8yUzJ5g04vX4Czla_VkFP3TiID-LJYkfXC5YZEFy1mZ82WBT9vIB7taTYNQShVq26YwpLCHV1s6BHYrSYYdlc
  }
}

[TokenCache] Using cached owner_key
Traceback (most recent call last):
  File "/var/tmp/GoogleFindMyTools/GoogleFindMyTools/main.py", line 9, in <module>
    list_devices()
  File "/var/tmp/GoogleFindMyTools/GoogleFindMyTools/NovaApi/ListDevices/nbe_list_devices.py", line 57, in list_devices
    get_location_data_for_device(selected_canonic_id)
  File "/var/tmp/GoogleFindMyTools/GoogleFindMyTools/NovaApi/ExecuteAction/LocateTracker/location_request.py", line 60, in get_location_data_for_device
    decrypt_location_response_locations(result)
  File "/var/tmp/GoogleFindMyTools/GoogleFindMyTools/NovaApi/ExecuteAction/LocateTracker/decrypt_locations.py", line 46, in decrypt_location_response_locations
    decrypted_location = decrypt_aes_gcm(identity_key_hash, encrypted_location)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/tmp/GoogleFindMyTools/GoogleFindMyTools/KeyBackup/cloud_key_decryptor.py", line 89, in decrypt_aes_gcm
    decrypted_data = aes_gcm.decrypt(iv, ciphertext, additional_data)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Nonce must be between 8 and 128 bytes
```

leonboe1, 2025-02-01:
I also experienced that sometimes Google does not send any location back - not even reports sent by an own device. However, I'm not sure if this is what happened here. The code crashes because the field `recentLocationTimestamp ` indicates that there should be a location in the `recentLocation` field, but the latter is empty. This should not happen. This could mean that a) there is a bug in Google's implementation, or b) there is a bug while parsing the response from the server.

If you want, you can send me the complete hex string from "Decoded FMDN Message: xxx" to my [email](mailto:mail@leonboettger.com), then I can take a look if there is a bug in the code.

Nonetheless, I added an additional check to prevent a crash from happening in this case.

leonboe1, 2025-02-02:
For everyone wondering:

The issue was the following: If you use Google Home and have your address added to your account, you might not get a coordinate for your tracker, but a semantic location, such as "Home". I added code that extracts this information.

closed 2025-02-02 by leonboe1

sistemistaitaliano, 2025-12-17:
> For everyone wondering:
> 
> The issue was the following: If you use Google Home and have your address added to your account, you might not get a coordinate for your tracker, but a semantic location, such as "Home". I added code that extracts this information.

Wich type of code you have used for extract the gps info?