leonboe1/GoogleFindMyTools #9: Document BLE advertisement parameters

biemster, 2025-02-08:
The ESP32 firmware has a couple parameters before the EID, as described here:
https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#advertised-frames

The current version has some differences with for example https://github.com/biemster/FindMy/blob/main/ESP32/findmy_google.py, are there experiments ongoing with for example the unwanted tracking protection mode?

leonboe1, 2025-02-09:
Hi there, I'm not sure if this is what you mean, but the only difference I see is Unwanted Tracking Mode and the hashed flags.

Unwanted Tracking Mode is always enabled since the current firmware does not support the [Bluetooth requests](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#operations) for enabling/disabling it. As the developer of the anti-stalking app AirGuard, I simply cannot support the permanent deactivation of Unwanted Tracking Mode.

biemster, 2025-02-09:
> Hi there, I'm not sure if this is what you mean,

yeah sorry the question was a bit unclear. I was wondering about exactly the two bytes you mention. Since you use the exact same flags as I initially picked I was wondering why you still include the hashed flags (as indicated in the length), although I did not see that calculated in your firmware.

Second I did not realize that the unwanted tracking mode is actually accompanied with a service, but I fully understand your reasoning why you don't want to set it disabled. Do you have an idea how this mode actually prevents tracking?

biemster, 2025-02-09:
@pvvx can I somehow persuade you to add this advertisement to your st17h66 KEY2 firmware like you did here https://github.com/pvvx/THB2/issues/94?

pvvx, 2025-02-09:
> @pvvx can I somehow persuade you to add this advertisement to your st17h66 KEY2 firmware like you did here https://github.com/pvvx/THB2/issues/94?

I will not support this resource and battery-hungry monstrosity.
This requires a special version of the device with a large battery.

biemster, 2025-02-09:
> I will not support this resource and battery-hungry monstrosity. This requires a special version of the device with a large battery.

really? I must be misunderstanding something here then.. I was under the impression that it was just an advertisement like the Apple FindMy, but with a keystone format and a specific EID as data payload. Why would this be more power intensive than the Apple advertisement?

pvvx, 2025-02-09:
https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn?#advertised-frames

The device must broadcast several advertising events every 2 seconds.

The Find My Device Network (FMDN) accessory specification does not support LE Long Range. As a result, it is a useless fiction for detecting surrounding devices located only a few meters away.

biemster, 2025-02-09:
> https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn?#advertised-frames
> 
> The device must broadcast several advertising events every 2 seconds.
> 
> The Find My Device Network (FMDN) accessory specification does not support LE Long Range.
> 
> In one room you will find such a device, but what if it is a farm plot?

yeah that is definitely not ideal, but also very similar to the Apple FindMy already implemented right? It would be nice to experiment and check how strict the 2 second interval will be enforced on the receiver side.

But most of all, how cool would it be if your firmware would be the first on the planet to support both the major FindMy networks? (I'm going to race you to beat you on this one actually)

pvvx, 2025-02-09:
> But most of all, how cool would it be if your firmware would be the first on the planet to support both the major FindMy networks?

Apple's Find My feature has not yet been implemented in any alternative project.
Broadcasting fixed BLE advertising in FindMy format is a non-functional stub, a mere pampering.

pvvx, 2025-02-09:
Here is the firmware with full implementation - https://pvvx.github.io/FindMyQMDv1/

biemster, 2025-02-09:
> Any project requires a lot of time to support users...

Very true, I spent 10x the time on user support for my FindMy projects than I do on development (I don't mind at all, but still)

pvvx, 2025-02-09:
> Very true, I spent 10x the time on user support for my FindMy projects than I do on development (I don't mind at all, but still)

In addition to user support, support is needed in popular applications. For example, in Home Assistant. But this is not yet available, and there are only scattered utilities that most users cannot handle. This is why FindMy is not developing.

leonboe1, 2025-02-11:
> yeah sorry the question was a bit unclear. I was wondering about exactly the two bytes you mention. Since you use the exact same flags as I initially picked I was wondering why you still include the hashed flags (as indicated in the length), although I did not see that calculated in your firmware.

The hashed flags are mot explicitly initialized (=0), but they are sent over the air because the length of the Bluetooth segment is explicitly set and includes them. Google's documentation mandates that they are sent with the advertisement if Unwanted Tracking Mode is enabled, that's why I decided to add them (even though it does not provide any additional information).

> edit: I forgot to mention how awesome this new feature is, this is a great win for your project! do you mind if I submit this to some tiplines like HaD?

Thanks! Sure, go ahead.

leonboe1, 2025-02-11:
> an idea, create a binary img with a byte pattern for adv_raw_data e.g. 23,24,25,26... and swap it with main.py, this eliminates the compilation process, so it's easier to try it out.

If you want this, feel free to create a PR.

biemster, 2025-02-12:
> > do you mind if I submit this to some tiplines like HaD?
> 
> Thanks! Sure, go ahead.

https://hackaday.com/2025/02/11/google-findmy-tools-run-on-an-esp32/

biemster, 2025-02-16:
Closing this as the initial point is addressed, and the conversation strayed quite a bit.

closed 2025-02-16 by biemster