leonboe1/GoogleFindMyTools #12: How to ring trackers

ItsHarper, 2025-02-09:
The README says "You can also send commands to your trackers over the network, such as ringing the tracker", but I couldn't find any details or code in the repo, besides the existence of a `ringing_key`. How would you do this?

leonboe1, 2025-02-11:
You are right, it's not included in the command line program as of now. The reason for this is that it is not that useful after all. Ringing trackers only work if an Android device logged in with your account is nearby. The code is located in NovaApi/ExecuteAction/PlaySound

ItsHarper, 2025-02-11:
Thanks, that's actually totally fine for my use case. Eventually I want to be able to tell Home Assistant via voice to ring my wallet or keys as I'm about to head out the door.

leonboe1, 2025-02-12:
I think for this it would be better to establish a Bluetooth connection to the tracker directly, as the network-based ringing can take several seconds.

closed 2025-02-12 by leonboe1

ItsHarper, 2025-02-13:
Do you know of any resources for doing that? 

FWIW, when I ring the device from the Find My app, it takes several seconds there too, unless I just rang it a moment before.

leonboe1, 2025-02-13:
You can use [this code](https://github.com/seemoo-lab/AirGuard-iOS/blob/main/Shared/Views/SheetView/Debug/FMDNOwnerOperations.swift) to understand how ringing works. It's also documented in Google's developer specification. You will need the Identity Key to derive the Ringing Key, but reading out the Identity Key is possible with my scripts.