leonboe1/GoogleFindMyTools #16: Can't run main.py

ZaleAnderson, 2025-02-12:
Howdy! Running this on Windows 11 with python 3.13 getting this error

Traceback (most recent call last):
  File "C:\Users\zalea\Desktop\GoogleFindMyTools-main\main.py", line 6, in <module>
    from NovaApi.ListDevices.nbe_list_devices import list_devices
  File "C:\Users\zalea\Desktop\GoogleFindMyTools-main\NovaApi\ListDevices\nbe_list_devices.py", line 7, in <module>
    from faulthandler import register
ImportError: cannot import name 'register' from 'faulthandler' (unknown location)

If you need any more info just let me know.

Eviltr0N, 2025-02-13:
Hi, I have also encountered same error, I am on windows 11 with Python 3.10.11

I can tell you a quick and dirty fix, edit `GoogleFindMyTools-main\NovaApi\ListDevices\nbe_list_devices.py` file and comment out the import statement `from faulthandler import register`, line 7.

Cheers!

leonboe1, 2025-02-13:
Thanks! Does this also work for you @ZaleAnderson ?

ZaleAnderson, 2025-02-13:
Good question. I gave up running this on windows and just installed Ubuntu instead which worked just fine. I went back to Windows and tried from scratch and never had this error again but still cannot get VSC to be happy.

closed 2025-02-13 by leonboe1