leonboe1/GoogleFindMyTools #20: [Feature req] Add OTA supporting

T-REX-XP, 2025-02-16:
Hi. It would be nice to add support OTA updates. Thanks

leonboe1, 2025-02-17:
Could you explain why this is helpful? I currently don't see any benefit since the firmware will not get updated frequently.

T-REX-XP, 2025-03-17:
It would be nice to avoid constantly using wires to burn the new FW. Also here is the good idea of using a few more services to setup the key via the browser [PHY62x2BTHome](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html) 

[image]

leonboe1, 2025-03-17:
> It would be nice to avoid constantly using wires to burn the new FW. Also here is the good idea of using a few more services to setup the key via the browser [PHY62x2BTHome](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html)
> 
> [image]

I don't really understand why you need to flash a new firmware so often?

T-REX-XP, 2025-03-18:
> > It would be nice to avoid constantly using wires to burn the new FW.
> 
> I don't really understand why you need to flash a new firmware so often?

I would like to have the ability to experiment with fw. So, it will require constantly deploying the fw. Did you experience with esp home OTA updates? It will be nice to gav the same here. No mo wires, just specific services to interact with using browser web Bluetooth for configuration. OTA support will significantly improve maintenance and feature experimenting.