leonboe1/GoogleFindMyTools #21: Issue registering device

SuperCartman, 2025-02-16:
I am having issues with Google. When I try to register a device I get an "You encrypted data is locked on this device" error message. I have set up Find My Device on my android device with the correct account but am still having issues. Does anyone have any ideas?

[image]

ASULTDGIT, 2025-02-17:
It seems google is forcefully expiring the token. May be detecting automation and forcing denial, or could be an issue with the original request.

leonboe1, 2025-02-17:
What OS are you using?

leonboe1, 2025-02-17:
One additional thing I noticed: "Your encryption data is locked on your device" is also shown if you have never paired a Find My Device tracker with an Android device. Have you tried to temporarily pair a tracker with your phone?

SuperCartman, 2025-02-17:
I was able to get it to work. I had to use an account that already had a device paired to it. At least that's what I think is going on. Thanks for your help

closed 2025-02-18 by leonboe1