leonboe1/GoogleFindMyTools #22: getting cryptography.exceptions.InvalidTag

P6g9YHK6, 2025-02-16:
trying 2 devices and getting these errors.

```
  ⚡YYYYY ❯❯ python main.py
Loading...

--------------------------------------------------
Welcome to GoogleFindMyTools!
--------------------------------------------------

The following trackers are available:
1. Xiaomi Mi 10T Lite: X
2. Chipolo ONE Point: X

If you want to see locations of a tracker, type the number of the tracker and press 'Enter'.
If you want to register a new ESP32- or Zephyr-based tracker, type 'r' and press 'Enter': 2
[LocationRequest] Requesting location data for device with canonic ID: 67adbd49-0000-2392-b563-582429c649f0
[FCMReceiver] Listening for notifications.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Location request successful.
[SharedKeyRetrieval] You need to log in again to access end-to-end encrypted keys to decrypt location reports.
> This script will now open Google Chrome on your device.
> Make that you allow Python (or PyCharm) to control Chrome (macOS only).

[SharedKeyRetrieval] Press 'Enter' to continue...
[SharedKeyFlow] Signed in successfully.
[SharedKeyFlow] Received Shared Key.
Traceback (most recent call last):
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\main.py", line 10, in <module>
    list_devices()
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ListDevices\nbe_list_devices.py", line 71, in list_devices
    get_location_data_for_device(selected_canonic_id)
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\location_request.py", line 58, in get_location_data_for_device
    decrypt_location_response_locations(result)
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\decrypt_locations.py", line 42, in decrypt_location_response_locations
    identity_key = retrieve_identity_key(device_registration)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\decrypt_locations.py", line 33, in retrieve_identity_key
    identity_key = decrypt_eik(owner_key, encrypted_identity_key)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\KeyBackup\cloud_key_decryptor.py", line 166, in decrypt_eik
    return decrypt_aes_gcm(owner_key, encrypted_eik)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\KeyBackup\cloud_key_decryptor.py", line 90, in decrypt_aes_gcm
    decrypted_data = aes_gcm.decrypt(iv, ciphertext, additional_data)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cryptography.exceptions.InvalidTag
Exception ignored in: <function Chrome.__del__ at 0x000001F518C6D6C0>
Traceback (most recent call last):
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] The handle is invalid
```

```
  ⚡YYYYY ❯❯ python main.py
Loading...

--------------------------------------------------
Welcome to GoogleFindMyTools!
--------------------------------------------------

The following trackers are available:
1. Xiaomi Mi 10T Lite: X
2. Chipolo ONE Point: X

If you want to see locations of a tracker, type the number of the tracker and press 'Enter'.
If you want to register a new ESP32- or Zephyr-based tracker, type 'r' and press 'Enter': 1
[LocationRequest] Requesting location data for device with canonic ID: 67adb9e9-0000-2e84-92bf-14223bbe07d6
[FCMReceiver] Listening for notifications.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Received response for a different request. Ignoring.
[FCMReceiver] Notification received!
[LocationRequest] Location request successful.
Traceback (most recent call last):
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\main.py", line 10, in <module>
    list_devices()
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ListDevices\nbe_list_devices.py", line 71, in list_devices
    get_location_data_for_device(selected_canonic_id)
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\location_request.py", line 58, in get_location_data_for_device
    decrypt_location_response_locations(result)
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\decrypt_locations.py", line 42, in decrypt_location_response_locations
    identity_key = retrieve_identity_key(device_registration)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\decrypt_locations.py", line 33, in retrieve_identity_key
    identity_key = decrypt_eik(owner_key, encrypted_identity_key)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\KeyBackup\cloud_key_decryptor.py", line 166, in decrypt_eik
    return decrypt_aes_gcm(owner_key, encrypted_eik)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\YYYYY\Downloads\GoogleFindMyTools-main\KeyBackup\cloud_key_decryptor.py", line 90, in decrypt_aes_gcm
    decrypted_data = aes_gcm.decrypt(iv, ciphertext, additional_data)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cryptography.exceptions.InvalidTag
```

leonboe1, 2025-02-17:
Can you please update to the latest version, go to folder Auth/secrets.json, open the file and remove this part at the end:
`, "owner_key": "XXX"`. Then, start main.py and please post the logging output again. Thanks.

P6g9YHK6, 2025-02-18:
updated now before removing:

```

 ⚡❯❯ py .\main.py
Loading...

--------------------------------------------------
Welcome to GoogleFindMyTools!
--------------------------------------------------

The following trackers are available:
1. Xiaomi Mi 10T Lite: 67adb9e9-0000-2e84-92bf-14223bbe07d6
2. Chipolo ONE Point: 67adbd49-0000-2392-b563-582429c649f0

If you want to see locations of a tracker, type the number of the tracker and press 'Enter'.
If you want to register a new ESP32- or Zephyr-based tracker, type 'r' and press 'Enter': 1
[LocationRequest] Requesting location data for Xiaomi Mi 10T Lite...
[FCMReceiver] Listening for notifications. This can take a few seconds...
[LocationRequest] Location request successful. Decrypting locations...
Failed to decrypt identity key encrypted with owner key version 2:
Traceback (most recent call last):
  File "C:\Users\x\Downloads\GoogleFindMyTools-main(1)\GoogleFindMyTools-main\main.py", line 10, in <module>
    list_devices()
  File "C:\Users\x\Downloads\GoogleFindMyTools-main(1)\GoogleFindMyTools-main\NovaApi\ListDevices\nbe_list_devices.py", line 72, in list_devices
    get_location_data_for_device(selected_canonic_id, selected_device_name)
  File "C:\Users\x\Downloads\GoogleFindMyTools-main(1)\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\location_request.py", line 56, in get_location_data_for_device
    decrypt_location_response_locations(result)
  File "C:\Users\x\Downloads\GoogleFindMyTools-main(1)\GoogleFindMyTools-main\NovaApi\ExecuteAction\LocateTracker\decrypt_locations.py", line 82, in decrypt_location_response_locations
    identity_key_hash = hashlib.sha256(identity_key).digest()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: object supporting the buffer API required
```

After removing same thing.

leonboe1, 2025-02-19:
Okay, I know what the problem is, but not sure how it can be fixed. Your "owner key" has version 2, which is something that never occurred to me.

Can you test the following: Go to the Auth folder, and delete the `secrets.json` file. Then, navigate to folder SpotApi/GetEidInfoForE2eeDevices, and open `get_eid_info_request.py`. Change this line

`get_eid_info_for_e2ee_devices_request.ownerKeyVersion = -1`

to

`get_eid_info_for_e2ee_devices_request.ownerKeyVersion = 2`

and try to run `main.py` again. If you still get the same error, delete `secrets.json` again, and try to change

`get_eid_info_for_e2ee_devices_request.hasOwnerKeyVersion = True`

to

`get_eid_info_for_e2ee_devices_request.hasOwnerKeyVersion = False`

in the same file.

Thanks!

leonboe1, 2025-02-19:
Edit: Now I know when this happens. If you forgot the PIN of your previous Android device, you can decide to reset end-to-end encrypted keys, which will increment the owner key version.

Did you by any chance do that in the past? And can you check if you see locations for this tracker in Google's Find My Device app?

P6g9YHK6, 2025-02-19:
i don't understand a word of what you said :D i do not use google services almost at all
my daily driver is pixel with grapheneOS and no google services 😅

the phone there for my test is a dedicated phone with an old google account with a fresh install of lineageos never did anything else than test your app with it and setup a tracker to test.

Test1 nothing modified:
```
PS C:\Users\YYYYY\Downloads\GoogleFindMyTools-main(2)\GoogleFindMyTools-main> py .\main.py
Loading...
GCM register request attempt 1 out of 100 has failed with Error=PHONE_REGISTRATION_ERROR
[FCMReceiver] Credentials updated.
[FCMReceiver] Listening for notifications. This can take a few seconds...
[AuthFlow] This script will now open Google Chrome on your device to login to your Google account.
> Please make sure that Chrome is installed on your system.
> For macOS users only: Make that you allow Python (or PyCharm) to control Chrome if prompted.

[AuthFlow] Press Enter to continue...
[AuthFlow] Installing ChromeDriver...
[ChromeDriver] Default ChromeDriver creation failed. Trying alternative paths...
[ChromeDriver] ChromeDriver started using C:\ProgramData\chocolatey\bin\chrome.exe
[AuthFlow] Waiting for 'oauth_token' cookie to be set...
[AuthFlow] Retrieved Account Token successfully.

--------------------------------------------------
Welcome to GoogleFindMyTools!
--------------------------------------------------

The following trackers are available:
1. Xiaomi Mi 10T Lite: 67adb9e9-0000-2e84-92bf-14223bbe07d6
2. Chipolo ONE Point: 67adbd49-0000-2392-b563-582429c649f0

If you want to see locations of a tracker, type the number of the tracker and press 'Enter'.
If you want to register a new ESP32- or Zephyr-based tracker, type 'r' and press 'Enter': 1
[LocationRequest] Requesting location data for Xiaomi Mi 10T Lite...
[LocationRequest] Location request successful. Decrypting locations...
[SharedKeyRetrieval] You need to log in again to access end-to-end encrypted keys to decrypt location reports.
> This script will now open Google Chrome on your device.
> Make that you allow Python (or PyCharm) to control Chrome (macOS only).

[SharedKeyRetrieval] Press 'Enter' to continue...
[ChromeDriver] Default ChromeDriver creation failed. Trying alternative paths...
[ChromeDriver] ChromeDriver started using C:\ProgramData\chocolatey\bin\chrome.exe
[SharedKeyFlow] Signed in successfully.
[SharedKeyFlow] Received Shared Key.
Retrieved owner key with version: 1

----------------------------------------
Attention:
----------------------------------------
Failed to decrypt identity key encrypted with owner key version 2, current owner key version is 1.
This may happen if you reset your end-to-end-encrypted data. To resolve this issue, open the folder 'Auth' and delete the file 'secrets.json'.
Exception ignored in: <function Chrome.__del__ at 0x0000023A1B48B240>
Traceback (most recent call last):
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] The handle is invalid
Exception ignored in: <function Chrome.__del__ at 0x0000023A1B48B240>
Traceback (most recent call last):
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] The handle is invalid
```

After deleting the secret and changing the value from -1 to 2:

```
[... 100 lines cut ...]
[SharedKeyFlow] Signed in successfully.
[SharedKeyFlow] Received Shared Key.
Retrieved owner key with version: 1

----------------------------------------
Attention:
----------------------------------------
Failed to decrypt identity key encrypted with owner key version 2, current owner key version is 1.
This may happen if you reset your end-to-end-encrypted data. To resolve this issue, open the folder 'Auth' and delete the file 'secrets.json'.
Exception ignored in: <function Chrome.__del__ at 0x0000020095F3F1A0>
Traceback (most recent call last):
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] The handle is invalid
Exception ignored in: <function Chrome.__del__ at 0x0000020095F3F1A0>
Traceback (most recent call last):
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\Users\YYYYY\AppData\Local\Programs\Python\Python312\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] The handle is invalid

```

leonboe1, 2025-02-20:
Okay. Can you go to the Find My Device app on your Xiaomi Mi 10T Lite, and see locations of the Chipolo ONE Point?

P6g9YHK6, 2025-02-20:
Yes.

P6g9YHK6, 2025-02-21:
> > Just to clarify a point can the software track both the phone and the tracker or only the tracker ?
> 
> It can do both. I'm not sure what went wrong with your account, but it seems like that the keys necessary to decrypt locations are only on your phone, and not uploaded to the cloud. You can try to change the Lock Screen PIN (yes, really), or just log out and log in again on your phone.

Done both and same result.

Matrix/Discord call to see if i am the idiot doing something wrong or there is maybe something intresting going on ?

si458, 2026-01-07:
just wondered if this has been resolved or a workaround at all?
i have the same problem and same error `InvalidTag`