leonboe1/GoogleFindMyTools #24: Add payload?

epe, 2025-02-19:
Hi! 
Thanks for your great work. would it be possible to publish the battery level as a payload. 
IIRC for Find my Devices you can not only see the location, but also its battery level. 

If i have a Powerbank, add an ESP32 that measures its capacity, can I add this value 0-100% somehow to this code to publish that ?

gsrbr, 2025-02-20:
Hello @epe ,
I don't know if Google FMDN supports a custom payload, but by the [Google FMDN docs](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn) it supports a simple battery status ussing the **Hashed flags** inside the _adv_data_raw_ , [Google FMDN Hashed flags docs](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#hashed-flags). 
**Supported battery status:**
- Battery level indication unsupported
- Normal battery level
- Low battery level
- Critically low battery level

Actually this repo has a simple and generic implementation of the "Firmware for custom ESP32-based trackers" so I think add it to the repo will be a little bit strange and will add complexity to the code.

I don't have how to test it now but i think you can add it just filling the **Hashed flags** inside the _adv_data_raw_

https://github.com/leonboe1/GoogleFindMyTools/blob/e2e0e430e011d233974082ae9385578117a7cc5f/ESP32Firmware/main/main.c#L60-L71

For add info on **Hashed flags** you can see this example
https://github.com/leonboe1/GoogleFindMyTools/blob/e2e0e430e011d233974082ae9385578117a7cc5f/ESP32Firmware/main/main.c#L72-L76

Remenber to respect the hashed flags field patern.

I hope it help :)

epe, 2025-02-20:
Oh, wow thanks for the insight.  Even a "feedback" of 4 states could probably even be helpful in some applications!

Further up in the document there is another chapter about battery notification:  https://developers.google.com/nearby/fast-pair/specifications/extensions/batterynotification?hl=de
Does this also apply for our usecase here or is that only feasible in another context?

leonboe1, 2025-02-20:
Just a note: Reading out the battery status is only possible when the tracker is nearby, it is not sent with network reports.

geissdoerfer, 2025-02-21:
> Oh, wow thanks for the insight. Even a "feedback" of 4 states could probably even be helpful in some applications!
>
> Further up in the document there is another chapter about battery notification: [...]

No this service is part of fast pair and basically a sibling to the FMDN functionality implemented in this repository.

sivar2311, 2025-02-22:
Sorry for this dumb question but how to calculate the hashed flags on the ESP32 ?
The docs says "To produce the final value of this byte, it is xor-ed with the least significant byte of SHA256(r)."
Is "SHA256(r)" the last byte of the EID ?
Or does the ESP32 have to have the secret key to perform this calculation?

leonboe1, 2025-02-22:
Yes, the EIK is needed for this, as well as key rotation. Both are not implemented, therefore you cannot implement the battery status as proposed by Google, but you could implement a custom, more simpler battery status advertisement.

sivar2311, 2025-02-22:
Could the EIK also exposed by main.py so this can be used no the ESP32 side?
Are you available for a chat like on discord or something?

epe, 2025-02-24:
I'm not so much into the technical Details. What is EIK/EID? It would be great if we could get this going .. there is plenty of usecases to send sensor data through this new kind of network :-)

leonboe1, 2025-02-24:
As mentioned above, the battery status is *not* sent over the network. Therefore, sending sensor data is not possible.

sivar2311, 2025-03-05:
@mrx23dot Check the [Google FMDN Hashed flags docs](https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn#hashed-flags).

The problem is you have to hash the flag byte. 
For this you have to use the EIK not the EID... (if i understand this correctly)

epe, 2025-03-05:
EID? EIK? 
If there is only 4 possible states, would it be possible to prepare the necessary crypto stuff and put it on the microcontroller, and that will just use one of for different values?