leonboe1/GoogleFindMyTools #25: Location not available?

epe, 2025-02-19:
Yeah, so i even managed to compile and flash this! 
I have had the Find my App enabled before and downloaded it. 
I see a new device and named it after my TTGO T7 microcontroller. 
It can't show the app, even tough I have an android phone here that is online and has Bluetooth and GPS enabled. 
What could be the problem? 

Did I read somewhere that the location can only be received with a python script or was that about openhay?

epe, 2025-02-19:
Oh, sure, there you go: 

In  ..../GoogleFindMyTools/ :
python3 main.py 
.. and then select the device from the menu.
 
Maybe add this to the README.me?

closed 2025-02-19 by epe

leonboe1, 2025-02-20:
What do you mean? The README says that locations can only be looked up with the Python script.

epe, 2025-02-20:
Couldnt find it here: https://github.com/leonboe1/GoogleFindMyTools

sivar2311, 2025-02-20:
@epe See 3rd point in "Known Issues" section here: [ESP32Firmware/README.md](https://github.com/leonboe1/GoogleFindMyTools/blob/main/ESP32Firmware/README.md#known-issues)

epe, 2025-02-20:
Yeah, I knew I had read something like that somewhere, but checking the main readme I couldn't find it.
It's a bit hidden and in my understanding it should be in the main README.md file since it affects the main.py and not specifically the Esp32 microcontroller code.

Thankful for your work!

leonboe1, 2025-03-06:
I will not move these issues to the main README, since they are specific to the firmware. However, I have added a link to those issues in the main README.

If you run the main.py after one week, you will still get reports that were collected AFTER running main.py. 

To my knowledge, not rotating the key is not a problem in the long term.

leonboe1, 2025-03-06:
> > If you run the main.py after one week, you will still get reports that were collected AFTER running main.py.
> 
> Interesting, I guess we tell the server to start collecting coordinates for the next 4days. Is it possible to send future date here? Like 4days from today, 4days from 4th day.. (since on the tag side nothing changes) It's just a weird design they don't store at least the last coordinate regardless.

I'm currently testing that.

> And is it enough to run main.py and exit, or I have to query position for each entry to start the 4days?

You don't have to query the position.