leonboe1/GoogleFindMyTools #26: Registered, built and deployed onto ESP32 - no location found

billybigpotatoes, 2025-02-21:
Result of Python Script

[DecryptLocations] Decrypted Locations:
No locations found.

Also getting Location unavailable error on phone.  - It keep respeating

Device is showing on both python list and phone application

Any thoughts?   Feels so close 

Built on mac using 5.4 or toolset

leonboe1, 2025-02-21:
As stated in the README, locations cannot be viewed in the Find My Device app.

Google's network is not the best, you need to be patient if you are located in a less-populated area.

leonboe1, 2025-02-21:
One additional thing: Did you use the ESP32 firmware or the Zepyhr firmware? I'm asking because the Zephyr firmware was not written by me, and I didn't have the time to test it yet.

billybigpotatoes, 2025-02-21:
Esp32 firmware.  Just checking the esp device supports latest ble. I am
getting errors trying to build for a esp32c6 board.

leonboe1, 2025-02-21:
Did you verify with a Bluetooth scanner app that the ESP32 is indeed sending out Bluetooth frames? nRF Connect is a good app for that, or AirGuard.

VaseSimion, 2025-03-03:
@billybigpotatoes C6 supports only BLE and not classic bluetooth, so the code won't work. 
I did a pull request that makes the code work for a C3 (because I have some at home on which I could test), and that should work for your C6 also.

leonboe1, 2025-03-03:
Have merged the PR, @billybigpotatoes does it work now?

billybigpotatoes, 2025-03-03:
I will give it it a test

adokitkat, 2025-03-06:
Hello, I am getting this:

```
3. GoogleFindMyTools µC: 67c74126-0000-287b-997d-94eb2c1af6e0

If you want to see locations of a tracker, type the number of the tracker and press 'Enter'.
If you want to register a new ESP32- or Zephyr-based tracker, type 'r' and press 'Enter': 3
[LocationRequest] Requesting location data for GoogleFindMyTools µC...
[FCMReceiver] Listening for notifications. This can take a few seconds...
[LocationRequest] Location request successful. Decrypting locations...
----------------------------------------
[DecryptLocations] Decrypted Locations:
No locations found.
```

I can see it in FindMyDevice app on the phone, but the location is unavailable there too. I can see it in nRF app as well.

leonboe1, 2025-03-06:
Did you verify that Find My Device works with regular trackers? Maybe it is not available in your area yet or there are too few devices.

billybigpotatoes, 2025-03-06:
Same

d

billybigpotatoes, 2025-03-07:
Rebuilt app on a standard ESP32 chip - logs are 

I (31) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (31) boot: compile time Feb 21 2025 00:48:59
I (31) boot: Multicore bootloader
I (33) boot: chip revision: v3.0
I (36) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1dd24h (122148) map
I (132) esp_image: segment 1: paddr=0002dd4c vaddr=3ff80000 size=0001ch (    28) load
I (132) esp_image: segment 2: paddr=0002dd70 vaddr=3ffbdb60 size=022a8h (  8872) load
I (139) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=78a4ch (494156) map
I (312) esp_image: segment 4: paddr=000a8a74 vaddr=3ffbfe08 size=02880h ( 10368) load
I (316) esp_image: segment 5: paddr=000ab2fc vaddr=40080000 size=17aa8h ( 96936) load
I (366) boot: Loaded app from partition at offset 0x10000
I (366) boot: Disabling RNG early entropy source...
I (376) cpu_start: Multicore app
I (384) cpu_start: Pro cpu start user code
I (385) cpu_start: cpu freq: 160000000 Hz
I (385) app_init: Application information:
I (385) app_init: Project name:     ESPFindMy
I (389) app_init: App version:      94faab7-dirty
I (393) app_init: Compile time:     Mar  7 2025 11:34:15
I (398) app_init: ELF file SHA256:  6b2fa3fad...
I (403) app_init: ESP-IDF:          v5.4-dirty
I (407) efuse_init: Min chip rev:     v0.0
I (411) efuse_init: Max chip rev:     v3.99
I (414) efuse_init: Chip rev:         v3.0
I (419) heap_init: Initializing. RAM available for dynamic allocation:
I (425) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (430) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (435) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (440) heap_init: At 3FFC7178 len 00018E88 (99 KiB): DRAM
I (445) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (451) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (456) heap_init: At 40097AA8 len 00008558 (33 KiB): IRAM
I (463) spi_flash: detected chip: generic
I (465) spi_flash: flash io: dio
W (468) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (481) coexist: coex firmware version: 49a8cdc
I (485) main_task: Started on CPU0
I (495) main_task: Calling app_main()
I (505) BTDM_INIT: BT controller compile version [20b459d]
I (505) BTDM_INIT: Bluetooth MAC: e0:5a:1b:74:df:b2
I (505) phy_init: phy_version 4840,02e0d70,Sep  2 2024,19:39:07
I (905) ESP_FMDN: Set BLE TX Power to 9 dBm
I (905) ESP_FMDN: BLE advertising started.
I (905) main_task: Returned from app_main()

No location found - but only just built - so will leave running for 24 hours