leonboe1/GoogleFindMyTools #28: Feature Req: main.py commandline parameters

epe, 2025-02-24:
Hi! 
I would suggest to add commandline parameters to the main.py? 

Something like: 

### Register
-r -e … register esp32 device
-r -z … register zephyr device 

### Locate
-l locate 
-i Tracker ID
-l -i 999999-0000-7777-4444-000000000 … locate tracker with certain id 

-d daemon mode
-d keep reporting location to stdout with a given interval

-f Format
Output format in json or directly for the traccar osmand format: https://www.traccar.org/osmand/

epe, 2025-03-15:
Not necessary, with commanline Parameters you can easily decide to run different tasks or interactive/non interactive modes..
A command line parameter is enough, just make a wrapper script and pipe the output to the right file or socket or port ..
Polling the last location once or in regular interval would be great

Paxy, 2025-04-01:
+1 for interval reporting for specific device