leonboe1/GoogleFindMyTools #45: [question] Use spot API to manually set location of device?

Tikolu, 2025-04-30:
I am trying to use this repo for an unusual use-case; I am getting the location of a device from a different API (polling every few seconds), and I would like to then send this location data to Google to update my device map.

I can use the API `https://spot-pa.googleapis.com/google.internal.spot.v1.SpotService/CreateBleDevice` to create a new device, and I can see it show up on my device list.
Is there a different API I could use to update the location of this device, similarly to how an Android device regularly sends its location?

Essentially, I want to completely bypass all of the BLE Advertising functionality and just manually update the location.

Tikolu, 2025-04-30:
I have multiple devices and tracking chips already on the Google Find My network, so it would be nice to have everything in one place.
Additionally, the sharing functionality is very useful and reliable (from my experience). By adding my custom tracking device to my Google account, I can easily share its location with others, and control who has access through the Find My app.

leonboe1, 2025-05-03:
Ah, I see. Yes, it is possible but a bit tedious to implement. I'll think about it.