leonboe1/GoogleFindMyTools #48: [question] possible to rename or give name for registering new device

fleshgordo, 2025-05-19:
I've registered two devices (one esp32 and one zephyr). Both appear as 
GoogleFindMyTools µC in the list. I can rename them only on the Android app. Any chance to rename them with main.py or set a name when registering a new device?

aaronjamt, 2025-05-20:
Copied from a comment I made on another related repo:
> Also, if you edit [SpotApi/CreateBleDevice/create_ble_device.py](https://github.com/leonboe1/GoogleFindMyTools/blob/97bba095e438563e499808bf4018ce76f60bbcd5/SpotApi/CreateBleDevice/create_ble_device.py), you can change the URL for the icon on [line 36](https://github.com/leonboe1/GoogleFindMyTools/blob/97bba095e438563e499808bf4018ce76f60bbcd5/SpotApi/CreateBleDevice/create_ble_device.py#L36) (I used http://static.tildacdn.com/tild6361-6634-4361-b433-396433333164/fpr-001_light_web_2_.png for my Flipper) and the manufacturer and model information on [lines 72-73](https://github.com/leonboe1/GoogleFindMyTools/blob/97bba095e438563e499808bf4018ce76f60bbcd5/SpotApi/CreateBleDevice/create_ble_device.py#L72C5-L73C38) (which I set to "Flipper" and "Flipper Zero"). If you want to get extra fancy, I also changed [line 31](https://github.com/leonboe1/GoogleFindMyTools/blob/97bba095e438563e499808bf4018ce76f60bbcd5/SpotApi/CreateBleDevice/create_ble_device.py#L31) to my Flipper's name, so that can also show up, although I believe you can also change that in the Find My app.

(https://github.com/MatthewKuKanich/FindMyFlipper/pull/115#issuecomment-2848722522)

That's targeted at using a Flipper Zero, but you can modify the values to whatever you want for your own device.

fleshgordo, 2025-05-20:
Perfect, thanks! I was wondering where the image of the ESP32 came from :)