leonboe1/GoogleFindMyTools #49: Build & Flash successul for ESP32c3 and nrf8211

fleshgordo, 2025-05-19:
I've followed instructions and installed the toolchain for both hardware (a XIAO ESP32-C3 and a nRF52811 that sells on Aliexpress as YJ-15044 beacon module).

ESP32-c3:
Changed the advertisement key as described. Adapted CMakeLists.txt
```
if(${IDF_TARGET} STREQUAL "esp32c3")
    set(SDKCONFIG_DEFAULTS "sdkconfig.defaults.esp32c3")
endif()
```

to enable Nimble for Bluetooth. Running the build with

```
idf.py -D SDKCONFIG_DEFAULTS=sdkconfig.defaults.esp32c3 set-target esp32c3
```
Could flash it succesfully with:

`idf.py -p /dev/tty.usbmodem121101 flash`

i can see it booting up and starting bluetooth stack as well with:
`idf.py monitor`

```
I (319) ESP_FMDN: Initializing BLE
I (319) BLE_INIT: BT controller compile version [d74042a]
I (319) BLE_INIT: Bluetooth MAC: 64:e8:33:8a:22:52
I (319) phy_init: phy_version 1200,2b7123f9,Feb 18 2025,15:22:21
I (369) ESP_FMDN: BLE Host Task Started
I (369) NimBLE: GAP procedure initiated: stop advertising.

I (369) NimBLE: Failed to restore IRKs from store; status=8

I (369) NimBLE: GAP procedure initiated: advertise;
I (369) NimBLE: disc_mode=2
I (379) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=32
I (389) NimBLE:

I (389) ESP_FMDN: Started advertising
I (389) ESP_FMDN: adv_raw_data:
02 01 06 19 16 aa fe 41
11 42 f3 4b 44 0c c7 f9
e5 d4 c5 98 ca cc d1 24
08 fd 5f f5 00 00 00
I (399) main_task: Returned from app_main()
````
However, i don't get any location updates from the device. I also can't see it in a Bluetooth scanner app such as "nRF Connect".

I have also compiled the Zephyr firmware for a nrf8211 tag (changed main.c and added my advertisement key)

```
west build -p -b nrf52840dk_nrf52840 app -d build
```

flashed it with a ST-LINK V2 usb dongle by using openocd

```
openocd -f interface/stlink.cfg -f target/nrf51.cfg \
  -c "adapter speed 100" \
  -c "program build/merged.hex verify reset exit"
```

Flashing was successful (output from open-ocd)
````
 ** Programming Started **
Info : nRF51822-QFAA(build code: H1) 256kB Flash, 16kB RAM
Info : Padding image section 0 at 0x00021774 with 12 bytes
Info : Padding image section 1 at 0x00024034 with 8 bytes
Info : Padding image section 2 at 0x00024693 with 1 bytes
Warn : Adding extra erase range, 0x00024924 .. 0x00024bff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

But same issue here. I can't see any of the tags in nRF Connect while scanning and FMD just shows "Location unavailable"

Any ideas how to further test/debug the tags?

fleshgordo, 2025-05-20:
Update: My bad for the nRF Connect app. The filter option was enabled, so the beacon tag got filtered out (it's the Exclude Google option). ESP32C3 shows up and I have received immediately a location. Will try the nrf81 tag later and give an update.

leonboe1, 2025-05-20:
Great 👍