leonboe1/GoogleFindMyTools #50: [question] how to extend battery life for ESP32?

fleshgordo, 2025-05-26:
Any ideas how to extend the battery life on ESPc3? 

I've changed the interval time to be not as fast:

```
.adv_int_min = 0x0800;  // 2048 × 0.625ms = 1.28 seconds
.adv_int_max = 0x0800;
```

But the ESP32 is still drawing around 20mA in idle mode. Did anyone else experiment with putting the ESP32 to deep sleep? 

Would it make sense to put the ESP32 to deep sleep every sec for example:

- Wake up every second
- Initialize BLE and advertise for a few packets
- Shut BLE down
- Return to deep sleep

leonboe1, 2025-05-26:
Did you already try to reduce the TX Power? It's currently set to the highest level.

kamo104, 2025-07-10:
@fleshgordo You can also take a look at [macless-openhaystack](https://github.com/dchristl/macless-haystack/blob/main/firmware/ESP32/src/openhaystack_main.c) which is apple's BLE tracking network implementation. They use light sleep there which might be transferable to this project as well.

biemster, 2025-07-25:
> I flashed d4 pico and it's power hungry enough not to be usable. Around 0.2W at all times. I don't even see slightest idle or sleep. This is too much, I was thinking to make cat's collar tracker on 60mAh battery but that would last less than an hour. Under month there is really no use.

ESP32 is not suitable for stuff like this, especially not on 60mAh. You'd want a wch ch5xx, telink tlsr825x or phy62x2 like what's explored in https://github.com/biemster/FindMy/issues/74.