leonboe1/GoogleFindMyTools #56: Overwriting the truncated Eids for devices

hallhu1, 2025-06-25:
I am trying to see if there is a way to change the EID for an existing device. I have been trying to use the "UploadPrecomputedPublicKeyIds" in the spot API but it doesn't seem to overwrite the truncated Eids on the google server. Any ideas for solving this?

leonboe1, 2025-06-28:
Hm, I think it is definitely possible to overwrite the truncated EIDs on the server. Just make sure that you are using timestamps which were already registered on the server. May I know what the goal is?

leonboe1, 2025-07-07:
I would say that it is completely normal if you still get reports from the old EID, since the last few reports are cached on the server.

Rolling keys (like in the real trackers) was already discussed, the issue with this is mainly that this depends on the time since the tracker was set up. Having a time counter on self-made trackers is difficult, since it would be reset if the power was cut.

Real Google trackers avoid this problem by a synchronization feature with the phone of the owner and Bluetooth, but this would need time to be implemented for the ESP32 firmware.

leonboe1, 2025-07-12:
Hm, that's weird. Maybe they are somewhat still cached? You could try to debug this by inspecting which keys are still saved on the server. You can retrieve them by a call to the Spot API (just like for registration, see the create_ble_device.py file) and to the  "ListEidsForBleDevices" endpoint with an empty payload. The response looks as follows: 
```
message ListEidsForBleDevicesResponse {
    repeated DevicePublicKeyIds deviceEids = 1;
    
    message DevicePublicKeyIds {
        CanonicId canonicId = 1;
        int32 rotationPeriod = 3;
        PublicKeyIdList serverList = 4;
    }
}
```

If you have time to tinker, you could try this out.