leonboe1/GoogleFindMyTools #58: Rotating Keys / Official App Support

aaronjamt, 2025-07-08:
What is holding this project back the most for me is the inability to use the official Google Find My Devices app/website. I'd like to help in any way I can, although I don't know nearly as much about this as you. What still needs to be done for this?

I know there is the rotating keys issue, since ESP32s don't keep time on reboot, but there are other devices which have RTCs or similar and I'd like to use this project with. How hard is this to implement, assuming you have a device with an accurate, reliable clock?

I know there is also the issue of Fast Pair being complicated. Is Fast Pair required for the official app, or since the Python script can register the tracker on your account, is that not needed?

I believe this was discussed previously but I can't find the thread on it, apologies if I've missed it!

leonboe1, 2025-07-08:
A fallback will be difficult, because the registration process differs for trackers without clock vs. with clock. So you need to device this beforehand.

About the library: What do you mean, abstract the Bluetooth level or the whole registration part?

aaronjamt, 2025-07-08:
> A fallback will be difficult, because the registration process differs for trackers without clock vs. with clock. So you need to device this beforehand.

I was meaning that it can be decided at compile-time, so the same code can be adapted for whatever RTC may be connected, or if none at all is.

About the library: What do you mean, abstract the Bluetooth level or the whole registration part?

I was meaning the code for the tracker specifically (the Bluetooth part). This would mean that the Zephyr and ESP32 parts could be somewhat combined--each would be a separate "project", but would include the "common" library part--as well as opening up the possibility of adding more platforms in the future. I'd like to help add a native Linux tracker, as well as fully integrate it into the [FindMy Flipper](https://github.com/MatthewKuKanich/FindMyFlipper) project for tracking a Flipper Zero.

For example, you could implement a new platform just by doing something like this:

```c
#include <platform-bluetooth.h>
#include "tracker.h"

uint32_t get_sys_time() {
    return platform_get_rtc_time();
}

void update_sys_time(uint32_t new_time) {
    platform_set_rtc_time(new_time);
}

Tracker tracker("TRACKER-KEY-FROM-PYTHON-SCRIPT", &get_sys_time, &update_sys_time);

int main() {
    platform_bluetooth_init();

    while (1) {
        platform_bluetooth_advertise(tracker.generate_advertisement_payload());
    }
}
```

leonboe1, 2025-07-08:
Sure, abstraction is always good, but don't try to spend too much time on it 😄

aaronjamt, 2025-07-08:
Let's say you had an ESP32 with a reliable, accurate, RTC. What actually needs to happen to make it work with the app? Is it just a matter of somehow "updating" the identifier in the advertisement, or is there something more that needs to be done there?

leonboe1, 2025-07-08:
The actual advertisement generation is not a huge issue. The only thing that might be a problem is the elliptic curve crypto you have to perform on the tracker. But it seems like the firmware provided by Nordic already implements that.

At the moment, I would try to see if the time sync is even possible. Otherwise, rotating advertisements would not make much sense.

As for the Google app, there's is actually not much more to do besides providing the rotating advertisement.

aaronjamt, 2025-07-08:
> The actual advertisement generation is not a huge issue. The only thing that might be a problem is the elliptic curve crypto you have to perform on the tracker. But it seems like the firmware provided by Nordic already implements that.

Sounds good.

> At the moment, I would try to see if the time sync is possible before implementing the advertisement generation. Otherwise, rotating advertisements would not make much sense. (But: If the Nordic firmware does not provide this time sync, we probably have to implement the advertisement generation before, because the tracker will not be recognized by the app otherwise).

Alright, I'll take a look. My phone doesn't have to have any special configuration for this, right?

> As for the Google app, there's is actually not much more to do besides providing the rotating advertisement.

Excellent!

aaronjamt, 2025-07-08:
Just from a cursory glance, the MessageStream looks like it can be just a single RFCOMM connection advertised with a fixed UUID (`df21fe2c-2515-4fdb-8886-f12c4d67927c`) and doesn't look too complicated.

leonboe1, 2025-07-12:
> 
> Alright, I'll take a look. My phone doesn't have to have any special configuration for this, right?

I don't think so, if you have the Find Hub installed registered with the same Google account as for the ESP32 trackers, testing should work.