leonboe1/GoogleFindMyTools #61: Not all devices are Discovered

xHecktor, 2025-07-17:
I have only a short question. 

in the android app find my device there are all devices:
Galaxy Note Pro 
Galaxy S25
Galaxy Tab A9+
Pixel Buds Pro
Chipolo Tag
Moto Tag

if I open the webbrowser with https://www.google.com/android/find/?login there are only:
Galaxy Note Pro 
Galaxy Tab A9+
Pixel Buds Pro
Galaxy S25

if I use now GoogleFindMyTools
Galaxy S25
Chipolo Tag
Moto Tag

are Discovered (witch is great). But why are  Galaxy Note Pro, Galaxy Tab A9+, Pixel Buds Pro missing? I am wondering because you use chrome, shouldn't the same devices be discovered as the webbrowser on windows?

As I told you, I am happy, because this are the important devices, I am only wondering why this is so

leonboe1, 2025-07-17:
Hello,

The Python script will only detect devices that are registered with Google's Find Hub network. Can you check if the network is really enabled on the devices that are missing?

leonboe1, 2025-07-21:
Yes, this looks normal, but you also stated that the S25 is detected by the Python script. Please check the same for a device you do not see In the Python script.

leonboe1, 2025-07-22:
> ## Welcome to GoogleFindMyTools!
> 
> The following trackers are available:
> 
> 1. Galaxy S25: 67b3b4e5-0000-29b3-8c34-30fd38173343
> 2. Jan Portemonnaie: 667db53d-0000-2eca-8777-883d24f6cc65
> 3. Jan Schlüssel: 667db623-0000-2265-ae86-883d24f96234
> 4. Naomi: 68766479-0000-26a7-aeb1-94eb2c1afa34
> 
> If you want to see locations of a tracker, type the number of the tracker and press 'Enter'. If you want to register a new ESP32- or Zephyr-based tracker, type 'r' and press 'Enter':
> 
> is that what you mean?

What I meant is to open the Find Hub app on a device you do not see in the Python script (such as the Galaxy Note Pro) and check there if the Find Hub network is enabled by clicking on the current device > Device Settings.

leonboe1, 2025-07-22:
> by the way, I modified the chrome.py a bit, because the log in did not work. I installed my own chrome driver and used the debug mode. You can find everything I did here: https://github.com/xHecktor/GoogleFindMyTools_for_raspberry-homeassistant/tree/main
> 
> I don't think it's the cause of the missing trackers, but maybe I'm wrong

Oh great, can you maybe open a Pull Request with those changes?

xHecktor, 2025-08-06:
@leonboe1 could you find something out?

O2Graphics, 2025-09-28:
I have a similar problem.
In Android "Find Hub" app, I'm can see all my devices: 

- Xiaomi phone (android 10)
- Lenovo tablet (android 6)
- "android PX5 car radio" (android 8)
- Several LocaTag trackers

With GoogleFindMyTools, I get almost every devices, even the tablet, except my PX5 car radio.

Like @xHecktor, in "NovaApi/ListDevices/nbe_list_devices.py" I have tried every values on the "wrapper.deviceListRequestPayload.type = DeviceUpdate_pb2.DeviceType.SPOT_DEVICE" line and I'm getting only my phone and tablet when the value is not SPOT_DEVICE.

I don't have any other devices to try.