leonboe1/GoogleFindMyTools #67: Failed to decrypt identity key encrypted with owner key version 2, current owner key version is 1.

Zwen23, 2025-08-19:
Hi,

i am getting the following error when retrieving a position for any of my devices.
I do get a list of my devices, when I choose one to request the position an error is displayed:

----------------------------------------
Attention:
----------------------------------------
Failed to decrypt identity key encrypted with owner key version 2, current owner key version is 1.
This may happen if you reset your end-to-end-encrypted data. To resolve this issue, open the folder 'Auth' and delete the file 'secrets.json'.
Exception ignored in: <function Chrome.__del__ at 0x000001C0F6E47420>
Traceback (most recent call last):
  File "C:\tmp\GoogleFindMyTools-homeassistant-main\venv\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\tmp\GoogleFindMyTools-homeassistant-main\venv\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] Das Handle ist ungültig
Exception ignored in: <function Chrome.__del__ at 0x000001C0F6E47420>
Traceback (most recent call last):
  File "C:\tmp\GoogleFindMyTools-homeassistant-main\venv\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\tmp\GoogleFindMyTools-homeassistant-main\venv\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] Das Handle ist ungültig

Any Idea how this can be prevented ? (Removing secrets.json an restarting leads to the same result).

Sven

diogo-madeira, 2025-10-24:
I have the same problem. Can you solve it? How?

scr4t3r, 2026-01-09:
same issue unfortunately. tried playing around with it but not luck: did the process with 2FA, had to enter device pin in browser in the end etc. repeating it multiple times results in no change.
not sure if relevant, but using macos