leonboe1/GoogleFindMyTools #69: Unable to run script - can't find Chrome.

Ken7382, 2025-09-19:
I've read through the instructions and tried on **three** different machines; all Windows 10 or 11 machines with the latest version of Chrome (that is checked to ensure it's up-to-date.)

- Chrome is in the usual spot in Windows (c:\Program Files\Google\Chrome\Application\Chrome.exe)
- I also made sure Python is the latest version (3.13.7) and that it's in the path.
- Ran with venv and without
- Tried with python and python3
- Installed requirements (as per script, using pip)
- Ran script inside of CMD that's running under admin privileges
- Turned off Windows Firewall entirely (just to be sure)

I'm at a loss - I looked inside the chrome_driver.py script and all the Windows paths have an "r" in front of them, wasn't sure what that's supposed to do (I'm not a Python programmer.) For grins, I took that out but it still didn't make a difference.

Might whip out my old Macbook and try that way?

Any help is appreciated!

[image]

There's a lot more to the errors but I figured it wasn't installing the chrome driver and that was the cause of the subsequent errors.

Ken7382, 2025-09-20:
Interesting - I noticed yours is under "programmes (x86)" (English equivalent is "Program Files (x86)") but mine is under "Program Files" (not the x86 version.) Either way, it _should_ detect it?

Idaho947, 2025-09-20:
See what you have in : chrome_driver.py

But I've the right path and it doesn't work :

[image]

Idaho947, 2025-09-20:
Sorry but don think a path issue. In both case the path are in the file but it doesn't work.
It don t work for Ken too.

Idaho947, 2025-09-21:
Unfortunately, it didn't work for me.

```
C:\WINDOWS\system32>cd C:\Users\mcham\Downloads\GoogleFindMyTools-main2\GoogleFindMyTools-main

C:\Users\mcham\Downloads\GoogleFindMyTools-main2\GoogleFindMyTools-main>python main.py
Loading...
[FCMReceiver] Credentials updated.
[FCMReceiver] Listening for notifications. This can take a few seconds...
[AuthFlow] This script will now open Google Chrome on your device to login to your Google account.
> Please make sure that Chrome is installed on your system.
> For macOS users only: Make that you allow Python (or PyCharm) to control Chrome if prompted.

[AuthFlow] Press Enter to continue...
[AuthFlow] Installing ChromeDriver...
[ChromeDriver] Default ChromeDriver creation failed: Message: session not created: cannot connect to chrome at 127.0.0.1:56081
from chrome not reachable; For documentation on this error, please visit: https://www.selenium.dev/documentation/webdriver/troubleshooting/errors#sessionnotcreatedexception
Stacktrace:
        GetHandleVerifier [0x0xae0c13+66051]
        GetHandleVerifier [0x0xae0c54+66116]
        (No symbol) [0x0x8bd980]
        (No symbol) [0x0x8b14fa]
        (No symbol) [0x0x8f7a36]
        (No symbol) [0x0x8eddcf]
        (No symbol) [0x0x8edbf6]
        (No symbol) [0x0x935c54]
        (No symbol) [0x0x93555a]
        (No symbol) [0x0x929d86]
        (No symbol) [0x0x8fb53e]
        (No symbol) [0x0x8fc414]
        GetHandleVerifier [0x0xd28a13+2457603]
        GetHandleVerifier [0x0xd239d2+2437058]
        GetHandleVerifier [0x0xb097f2+232930]
        GetHandleVerifier [0x0xaf9a18+167944]
        GetHandleVerifier [0x0xb0092d+196381]
        GetHandleVerifier [0x0xae8ee8+99544]
        GetHandleVerifier [0x0xae9082+99954]
        GetHandleVerifier [0x0xad322a+10266]
        BaseThreadInitThunk [0x0x761cfcc9+25]
        RtlGetAppContainerNamedObjectPath [0x0x773282ae+286]
        RtlGetAppContainerNamedObjectPath [0x0x7732827e+238]

[ChromeDriver] Trying alternative paths...
[ChromeDriver] ChromeDriver failed using path C:\Program Files (x86)\Google\Chrome\Application\chrome.exe: Message: session not created: cannot connect to chrome at 127.0.0.1:56182
from chrome not reachable; For documentation on this error, please visit: https://www.selenium.dev/documentation/webdriver/troubleshooting/errors#sessionnotcreatedexception
Stacktrace:
        GetHandleVerifier [0x0x3e0c13+66051]
        GetHandleVerifier [0x0x3e0c54+66116]
        (No symbol) [0x0x1bd980]
        (No symbol) [0x0x1b14fa]
        (No symbol) [0x0x1f7a36]
        (No symbol) [0x0x1eddcf]
        (No symbol) [0x0x1edbf6]
        (No symbol) [0x0x235c54]
        (No symbol) [0x0x23555a]
        (No symbol) [0x0x229d86]
        (No symbol) [0x0x1fb53e]
        (No symbol) [0x0x1fc414]
        GetHandleVerifier [0x0x628a13+2457603]
        GetHandleVerifier [0x0x6239d2+2437058]
        GetHandleVerifier [0x0x4097f2+232930]
        GetHandleVerifier [0x0x3f9a18+167944]
        GetHandleVerifier [0x0x40092d+196381]
        GetHandleVerifier [0x0x3e8ee8+99544]
        GetHandleVerifier [0x0x3e9082+99954]
        GetHandleVerifier [0x0x3d322a+10266]
        BaseThreadInitThunk [0x0x761cfcc9+25]
        RtlGetAppContainerNamedObjectPath [0x0x773282ae+286]
        RtlGetAppContainerNamedObjectPath [0x0x7732827e+238]
        (No symbol) [0x0]
[... 124 lines cut ...]
    driver = create_driver()
  File "C:\Users\mcham\Downloads\GoogleFindMyTools-main2\GoogleFindMyTools-main\chrome_driver.py", line 100, in create_driver
    raise Exception(
    ...<3 lines>...
    )
Exception: [ChromeDriver] Failed to install ChromeDriver. A current version of Chrome was not detected on your system.
If you know that Chrome is installed, update Chrome to the latest version. If the script is still not working, set the path to your Chrome executable manually inside the script.
Exception ignored in: <function Chrome.__del__ at 0x0000025A521BE980>
Traceback (most recent call last):
  File "C:\Users\mcham\AppData\Local\Programs\Python\Python313\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
    self.quit()
  File "C:\Users\mcham\AppData\Local\Programs\Python\Python313\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
    time.sleep(0.1)
OSError: [WinError 6] Descripteur non valide
Exception ignored in: <function Chrome.__del__ at 0x0000025A521BE980>
Traceback (most recent call last):
  File "C:\Users\mcham\AppData\Local\Programs\Python\Python313\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
  File "C:\Users\mcham\AppData\Local\Programs\Python\Python313\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
OSError: [WinError 6] Descripteur non valide
Exception ignored in: <function Chrome.__del__ at 0x0000025A521BE980>
Traceback (most recent call last):
  File "C:\Users\mcham\AppData\Local\Programs\Python\Python313\Lib\site-packages\undetected_chromedriver\__init__.py", line 843, in __del__
  File "C:\Users\mcham\AppData\Local\Programs\Python\Python313\Lib\site-packages\undetected_chromedriver\__init__.py", line 798, in quit
OSError: [WinError 6] Descripteur non valide
```

kllngtme, 2025-11-13:
@BSkando , I tried your update as well with no luck. Verified Chrome is up to date and the path is correct. Another fyi - your discord link doesn't work in your repo..